treehouses/cli #60: test treehouses bridge

i5o, 2018-05-19:


i5o, 2018-05-19:
Rpi Zero W
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword** **HostpotPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **""** **HotspotPassword**_

i5o, 2018-05-19:
Rpi Zero W USB
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword** **HostpotPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **""** **HotspotPassword**_

i5o, 2018-05-19:
Rpi Zero 3B
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword** **HostpotPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **""** **HotspotPassword**_

i5o, 2018-05-19:
Rpi Zero 3B+
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword** **HostpotPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **""** **HotspotPassword**_

i5o, 2018-05-19:
Rpi 3B+
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword** **HostpotPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **""** **HotspotPassword**_

i5o, 2018-05-19:
Rpi 3B+ eth0
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword** **HostpotPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **NetworkPassword**_
- [ ] treehouses bridge _**NetworkName** **HostpotName**_
- [ ] treehouses bridge _**NetworkName** **HostpotName** **""** **HotspotPassword**_

i5o, 2018-05-19:
**_testing rpi3b+ notes_**
```
sudo treehouses upgrade
sudo raspi-config  # set wifi country to 'US' (4, 4, us)
sudo treehouses bridge IgnacioGuest 3bplusrpi
sudo reboot
```

xyb994, 2018-06-01:
Looks like Rpi Zero W usb0 all works, have to wait a little longer before and after reboot...

closed 2018-06-20 by dogi