treehouses/cli #118: Add LED status help commands in CLI

rhuturajm, 2018-07-02:
There are different raspi versions having different LED configurations (1 or 2 LEDs with a status LED (PWR) common to all rpi boards) such as: (PWR: POWER, ACT: Activity)
RPI 3 B+ - 2 LEDs - PWR , ACT
RPI 3 - 2 LEDs - PWR , ACT
RPI Zero - 1 LEDs - ACT
RPI 2 - 2 LEDs - PWR , ACT 
RPI A+ - 2 LEDs - PWR , ACT
RPI B+ - 2 LEDs - PWR , ACT
RPI A - 2 LEDs - PWR , ACT
RPI B (rev 2) - 5 LEDs - ACT, PWR, FDX, LNK and 100
RPI B (rev 1) - 5 LEDs - OK, PWR, FDX, LNK and 10M

The LEDs have the following meanings :

| Label        | LED color           | Meaning  |
|:-------------: |:-------------:| :-----:|
| OK/ACT | Green | SD card activity |
| PWR | Red | Power |
| FDX | Green | Full Duplex (LAN) connected |
| LNK | Green | Link/Activity (LAN) |
| 10M/100 | Yellow | 100Mbit (LAN) connected |

(For later RPI models such as Pi 3, Pi 2 and B+ Ethernet socket has two additional status LEDs.)

| Label        | LED color           | Meaning  |
|:-------------: |:-------------:| :-----:|
| 10M/100 | Yellow | 100Mbit (LAN) connected |
| LNK | Green | Link/Activity (LAN) |

dogi, 2018-07-12:
@i5o look at https://github.com/ole-vi/pirateship/issues/9

i5o, 2018-07-12:
```
root@raspberrypi:/sys/devices/platform/leds/leds/led0# cat trigger 
none rc-feedback kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock timer oneshot [heartbeat] backlight gpio cpu cpu0 cpu1 cpu2 cpu3 default-on input panic mmc1 mmc0 rfkill-any rfkill0 rfkill1 
root@raspberrypi:/sys/devices/platform/leds/leds/led0# 
```

i5o, 2018-07-12:
```
none                No trigger
kbd-scrolllock      Keyboard scroll lock
kbd-numlock         Keyboard num lock
kbd-capslock        Keyboard caps lock
kbd-kanalock        Keyboard kana lock
kbd-shiftlock       Keyboard shift
kbd-altgrlock       Keyboard altgr
kbd-ctrllock        Keyboard ctrl
kbd-altlock         Keyboard alt
kbd-shiftllock      Keyboard left shift
kbd-shiftrlock      Keyboard right shift
kbd-ctrlllock       Keyboard left ctrl
kbd-ctrlrlock       Keyboard right ctrl
timer               Flash at 1 second intervals
oneshot             Flash only once
heartbeat           Flash like a heartbeat (1-0-1-00000)
backlight           Always on
gpio                Flash when a certain GPIO is high???
cpu0                Flash on cpu0 usage
cpu1                Flash on cpu1 usage
cpu2                Flash on cpu2 usage
cpu3                Flash on cpu3 usage
default-on          Always on
[input]             Default state
panic               Flash on kernel panic
mmc0                Flash on mmc0 (primary SD Card interface) activity
mmc1                Flash on mmc1 (secondary SD Card interface) activity
rfkill0             Flash on wifi activity
rfkill1             Flash on bluetooth activity
```
https://raspberrypi.stackexchange.com/questions/69674/are-there-other-act-led-trigger-options-besides-mmc-and-heartbeat/69734

rhuturajm, 2018-07-18:
@dogi Should I create a individual issue for each trigger and explain the trigger in detail?

i5o, 2018-07-19:
I think this issue should be merged with 
https://github.com/treehouses/builder/issues/74

otherwise it makes no sense to show what each trigger means