treehouses/cli #349: treehouses tor illegally twice same external port?

lillyxxx, 2019-08-02:
```
root@treehouses:~# treehouses tor add 22
Success: the port has been added
root@treehouses:~# treehouses tor list
external <=> local
22 <=> 22
root@treehouses:~# treehouses tor add 2222 22
Success: the port has been added
root@treehouses:~# treehouses tor list
external <=> local
22 <=> 2222
22 <=> 22
```

lillyxxx, 2019-08-02:
The good thing: if I do the same thing with the local port its automatically fixing itself
```
root@treehouses:~# treehouses tor add 22
Success: the port has been added
root@treehouses:~# treehouses tor add 22 2222
Success: the port has been added
root@treehouses:~# treehouses tor list
external <=> local
2222 <=> 22
```

dogi, 2019-08-11:
https://github.com/treehouses/cli/pull/367 fixes it

closed 2019-08-11 by dogi