treehouses/cli #371: Improve npmjs code quality

paulbert, 2019-08-12:
### The Problem

Currently the quality score of the CLI in npm is stuck at 80%:
[image]
(from: https://www.npmjs.com/search?q=treehouses)

Digging into the npms.io quality score that is used here we see **tests** as one of the main things we could improve to increase this score.
```
"evaluation": {
    "quality": {
      "carefulness": 0.71,
      "tests": 0.25,
      "health": 1,
      "branding": 0.15
    },
    "popularity": {
      "communityInterest": 57,
      "downloadsCount": 941,
      "downloadsAcceleration": 2.6456240487062415,
      "dependentsCount": 0
    },
    "maintenance": {
      "releasesFrequency": 1,
      "commitsFrequency": 1,
      "openIssues": 1,
      "issuesDistribution": 0.7075747518362785
    }
  },
```

### Solution

There needs to first be some research into how npms analyzes code quality to ensure the library used for unit tests or other testing will work well to improve this score (see repo here: https://github.com/npms-io/npms-analyzer)

Once we've identified a workable library for this, we can then implement some tests.

Liew211, 2019-08-24:
https://github.com/npms-io/npms-analyzer/blob/master/lib/analyze/evaluate/quality.js

LordJashin32, 2019-12-15:
[AGPL-3.0](https://spdx.org/licenses/) is a deprecated license. [Tests](https://npms.io/about) were third on the list for upping the quality score. The Read me file i think needs to be updated as well to boost the popularity (links, etc). I'm not sure what you guys think about testing but [bats](https://www.npmjs.com/package/bats) looks interesting. I'm sure all these shell scripts are as fast as they get but it would still be fun to time each one of them in production to see whats fast and slow.

closed 2020-02-29 by dogi