treehouses/cli #418: treehouses tor add check second number

hiroTochigi, 2019-09-02:


Kid243, 2019-09-05:
root@Kid243Pi4:~/cli# ./cli.sh tor add 8080 80
sed: -e expression #1, char 35: unterminated `s' command
Success: the port has been added

Kid243, 2019-09-05:
```
root@Kid243Pi4:~/cli# ./cli.sh tor list
external <=> local
80  <=> 8080
2200  <=> 2200
22  <=> 22
80  <=> 80
```

Kid243, 2019-09-05:
```
root@Kid243Pi4:~/cli# ./cli.sh tor add 8080 80
Success: the port has been added
root@Kid243Pi4:~/cli# ./cli.sh tor notice now
Thanks for the feedback!
root@Kid243Pi4:~/cli# ./cli.sh tor list
external <=> local
8080  <=> 80
root@Kid243Pi4:~/cli# ./cli.sh tor add 80
Success: the port has been added
root@Kid243Pi4:~/cli# ./cli.sh tor list
external <=> local
80  <=> 80
8080  <=> 80
root@Kid243Pi4:~/cli# ./cli.sh tor add 80 8080
Success: the port has been added
root@Kid243Pi4:~/cli# ./cli.sh tor list
external <=> local
80  <=> 80
80  <=> 8080
```

commit f666e489 (2019-09-05): treehouses tor add checks the 2nd number (fixes #418)
commit c65cfe13 (2019-09-07): treehouses tor add local port validation (fixes #418) (#421)
closed 2019-09-07 by dogi