treehouses/cli #424: treehouses wifi should not work

lillyxxx, 2019-09-05:
```
root@treehouses:/home/pi# treehouses wifi
^C^C^C^C^C^C^C^[[A^[[A^[[A^[[A^[[B^[[B^[[B^C^C^C
open wifi network
```
it seems that the command `treehouses wifi` try to connect a wifi with empty name.

lillyxxx, 2019-09-05:
also
```
root@treehouses:/home/pi# treehouses help wifi

Usage: treehouses wifi <ESSID> [password]

Connects to a wifi network

Example:
 treehouses wifi home homewifipassword
     Connects to a wifi network named 'home' with password 'homewifipassword'.

 treehouses wifi yourwifiname
     Connects to an open wifi network named 'home'.
```
the command `treehouses wifi yourwifiname` means `Connects to an open wifi network named 'yourwifiname'.` not `'home'`

irisb1701, 2019-09-09:
wifi.sh contains condition for no input 
```
if [ -z "$1" ]; then
    echo "Error: name of the network missing"
    exit 1
  fi
```
This works when testing with ./cli.sh
```
./cli.sh wifi
Error: name of the network missing
```
When using `treehouses wifi`
on rpi remote it disconnects pi
on rpi linked with ethernet it freezes like shown in issue description
on vagrant, it should return `Error: Must be run with rpi system`
but instead returns:

```
root@planet:/home/vagrant/cli# treehouses wifi
cp: cannot create regular file '/lib/dhcpcd/dhcpcd-hooks/10-wpa_supplicant': No such file or directory
/usr/lib/node_modules/@treehouses/cli/modules/wifi.sh: line 3: /etc/wpa_supplicant/wpa_supplicant.conf: No such file or directory
/usr/lib/node_modules/@treehouses/cli/modules/wifi.sh: line 3: /etc/wpa_supplicant/wpa_supplicant.conf: No such file or directory
open wifi network
```

hiroTochigi, 2019-09-14:
@dogi 
Please tell us from which version the wifi validation was implemented. 
@lillyxxx @irisb1701 
Could you try to break the latest image by using treehouses wifi?

dogi, 2019-09-16:
best way is using `treehouses remote` when you will loose rpi during process

phamduchongan93, 2019-09-19:
@irisb1701 @Kid243 @hiroTochigi work on my end with treehosues image 109.
```
pi@treehouses:~/cli$ sudo ./cli.sh wifi
Error: name of the network missing
pi@treehouses:~/cli$ sudo treehouses wifi
Error: name of the network missing
pi@treehouses:~/cli$ 
```
cli was pulled on Thur 19 2019

lillyxxx, 2019-09-19:
on treehouses-109 version, there's no such problem.

closed 2019-09-19 by lillyxxx