treehouses/cli #431: How to connect to WPA2/PEAP/MSCHAPv2 enterprise wifi networks

phamduchongan93, 2019-09-16:
Find a solution to  connect rp4 to enterprise wifi networks.
FYI:
> in ubuntu base, This following article mentions where they contain for the rp4  [How to connect to WPA2/PEAP/MSCHAPv2 enterprise wifi networks](https://askubuntu.com/questions/279762/how-to-connect-to-wpa2-peap-mschapv2-enterprise-wifi-networks-that-dont-use-a-c#295600).
> This [network manager](https://raspberrypi.stackexchange.com/questions/29783/how-to-setup-network-manager-on-raspbian) utility support WPA2/PEAP/MSCHAPv2 enterprise wifi networks.

dogi, 2019-09-16:
https://gist.github.com/elec3647/1e223c02ef2a9a3f836db7984011b53b

dogi, 2019-11-07:
old https://www.raspberrypi.org/forums/viewtopic.php?t=111100

dogi, 2019-11-07:
https://www.raspberrypi.org/forums/viewtopic.php?t=44029

LordJashin32, 2020-05-18:
https://github.com/CompEng0001/RaspberryPiEnterpriseSetup/blob/master/wpa_enterprise.sh

commit 9e3be473 (2020-06-08): `treehouses wifi` & `wifihidden` enterprise (fixes #431) (#1290)
closed 2020-06-08 by dogi