treehouses/cli #432: treehouses clone/backup

dogi, 2019-09-16:
... check if we can use this commands

vmnet8, 2019-09-17:
I tried this command on release-dockerexp1, it worked. 
```
root@vmnet8:~# treehouses clone /dev/sda
copying....
122112+0 records in
122112+0 records out
128043712512 bytes (128 GB, 119 GiB) copied, 5518.69 s, 23.2 MB/s
```

CalebProvost, 2019-09-19:
After wiping my USB I was able to get the clone to work.
```
root@treehouses:~# treehouses clone /dev/sda
copying....
14772+0 records in
14772+0 records out
15489564672 bytes (15 GB, 14 GiB) copied, 2073.35 s, 7.5 MB/s
```

And then verified `rebootneeded`
```
root@treehouses:~# treehouses rebootneeded
true
```

Everything appears to work. Even when running the command correctly `treehouses clone /dev/sda` without the following `/`
😄

closed 2019-09-19 by phamduchongan93

vmnet8, 2019-09-19:
I got different results with yours after exectuing clone command, even thought the OS is already read -only. 
```
root@vmnet8:~# touch a.txt
touch: cannot touch 'a.txt': Read-only file system
root@vmnet8:~# treehouses rebootneeded
false
```

vmnet8, 2019-09-22:
This time I used 32G SD card to clone and added some codes in clone.sh to echo the total seconds (2187 seconds), almost the same as the dd command shows (2186.25s).
```
pi@pi:~/git_repo/cli $ sudo ./cli.sh clone /dev/sda
copying....
30145+0 records in
30145+0 records out
31609323520 bytes (32 GB, 29 GiB) copied, 2186.25 s, 14.5 MB/s
Done. It took 2187 seconds to complete cloning.
```