treehouses/cli #433: treehouses speedtest?

dogi, 2019-09-16:


dogi, 2019-09-16:
- ping
- tracerout
- fast.com? and others ;)
  (speedtest.net)

vmnet8, 2019-09-23:
I found speedtest-cli package and installed it on Pi, and ran speedtest
```
pi@pi:~ $ sudo apt install speedtest-cli
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  python3-pyperclip rpi.gpio-common
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  speedtest-cli
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 23.3 kB of archives.
After this operation, 102 kB of additional disk space will be used.
Get:1 http://mirrors.ocf.berkeley.edu/raspbian/raspbian buster/main armhf speedtest-cli all 2.0.2-1 [23.3 kB]
Fetched 23.3 kB in 1s (26.6 kB/s)        
Selecting previously unselected package speedtest-cli.
(Reading database ... 151737 files and directories currently installed.)
Preparing to unpack .../speedtest-cli_2.0.2-1_all.deb ...
Unpacking speedtest-cli (2.0.2-1) ...
Setting up speedtest-cli (2.0.2-1) ...
Processing triggers for man-db (2.8.5-2) ...

pi@pi:~ $ speedtest
Retrieving speedtest.net configuration...
Testing from Comcast Cable...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Tekify Fiber & Wireles [18.93 km]: 18.432 ms
Testing download speed................................................................................
Download: 69.91 Mbit/s
Testing upload speed......................................................................................................
Upload: 2.46 Mbit/s
```
Can we include this `speedtest-cli` in our image, then we can implement a new cli module `speedtest` which simply run `/usr/bin/speedtest` command?

vmnet8, 2019-09-28:
beside adding a speedtest function, what argument is supposed to pass to `speedtest` command?

commit 1bba16e0 (2019-10-07): treehouses speedtest (fixes #433) (#444)
closed 2019-10-07 by dogi