treehouses/cli #467: sshkey accepts any input and deletes any key that matches input

irisb1701, 2019-10-16:
```
root@treehouses:~/cli# ./cli.sh sshkey add "hello"
====== Added to 'pi' and 'root' user's authorized_keys ======
hello
root@treehouses:~/cli# ./cli.sh sshkey list
==== root keys ====
hello
==== pi keys ====
hello
root@treehouses:~/cli# ./cli.sh sshkey delete "h"
h is deleted.
root@treehouses:~/cli# ./cli.sh sshkey list
==== root keys ====
==== pi keys ====
root@treehouses:~/cli# 

```

kjong, 2019-11-07:
It seems like this issue was fixed with PR #490?

kjong, 2019-11-07:
Is there any point in keeping the `sshkey delete` function when `sshkey github deleteuser` was added?

If yes, I'll open a PR with the changes to make `sshkey delete` correctly match the input.

PR #500 (2019-12-14): treehouses sshkey delete partial match (fixes #467)
Fixes #467
commit b0c8138c (2019-12-14): treehouses sshkey delete partial match (fixes #467) (#500)
closed 2019-12-14 by dogi