treehouses/cli #475: SSH Login Message or message board

phamduchongan93, 2019-10-19:
Desgin a message board to prompt user about tree-houses image. This message will appear when user ssh to the raspberry pi.

```
Linux li234-205 3.6.5-x86_64-linode28 #1 SMP Sun Nov 4 12:40:43 EST 2012 x86_64
Last login: Sat Dec 22 13:38:52 2012 from crazy-hostname

 16:07:17 up  2:36,  1 user,  load average: 0.00, 0.01, 0.05
USER     TTY      FROM              LOGIN@   IDLE   JCPU   PCPU WHAT
vagrant pts/0    97-85-36-82.dhcp 16:07    0.00s  0.16s  0.00s w

Filesystem            Size  Used Avail Use% Mounted on
/dev/hda1              24G  1.1G   22G   5% /

Start by typing treehouses help to begin
```

or 

`treehouses intro`
```

Welcome to the treehouses project!

The Open Learning Exchange is a social benefit and for-purpose organization based in Cambridge, Massachusetts. Our mission is to provide universal quality education using open source materials and technology to overcome educational barrier
For more instruction, please visit.
http://treehouses.io/cli/

To begin usage, type treehouses help
```

vmnet8, 2019-10-21:
Do you mean add more info on /etc/motd ?

phamduchongan93, 2019-10-21:
@vmnet8 yup more precisely.

- /etc/issue
- /etc/issue.net
- /etc/motd

pattanawadee88, 2019-10-21:
why do you think it necessary? 
I do not quite understand.

phamduchongan93, 2019-10-21:
@pattanawadee88 not really a must, but It's cool to inform users. I knew this from sysadmins people.

pattanawadee88, 2019-10-21:
> @pattanawadee88 not really a must, but It's cool to inform users.

haha I see. So there is nothing wrong in what we having right now :)