treehouses/cli #504: treehouses sshkey add "ssh"

pattanawadee88, 2019-11-12:
```
root@treehouses:~# treehouses sshkey add "ssh" 
====== Added to 'pi' and 'root' user's authorized_keys ======
ssh
root@treehouses:~# treehouses sshkey list
==== root keys ====
ssh
==== pi keys ====
ssh
```

xavierh93, 2019-11-14:
What is the issue here?

irisb1701, 2019-11-14:
> What is the issue here?
@xavierh93 it looks like the command `sshkey add` will add anything you input to the sshkey list regardless if its even an sshkey or not

phamduchongan93, 2019-11-14:
I was thinking should we opt to add a file as sshkey instead of a regular argument. User can have a  text file which has the private key. After `treehouses sshkey add -f <sshkey.txt>` command executed, they should have their key in the authorized file.