treehouses/cli #527: `treehouses bridge` command fail

phamduchongan93, 2019-12-02:
I perform network installation over the treehouses remote app.
I can assign WIFI network and password for the Wifi. After that, I was able to turn on the ap(access point), but the ap doesn't provide the internet connection.

Theses are the command's report that I pulled up from the raspbery pi. Also, the version of image is **114**
Please visit my link to see the log.
https://gist.github.com/phamduchongan93/69c0ab996e980608d3e1286f8848f8dc

LordJashin32, 2020-01-02:
The wifi can have 2 modes. Its either connected to router to provide internet to computer. Or its hotspot to connect to for your phone. It cannot do both. If you want Internet + hotspot you need physical wired internet plugged into Pi, along with hotspot. 

As far as I know this command works fine. Please let me know if you try again.

phamduchongan93, 2020-01-03:
@LordJashin32 Thanks for bringing this up. I'm glad we are in the same ship on this issue. I can demo what I did on this Friday. I can know what I did wrong in real time. This feature is something I want to have.

closed 2020-02-29 by dogi