treehouses/cli #547: treehouses memory has no output

dogi, 2019-12-21:


dogi, 2019-12-21:
[image]

dogi, 2019-12-21:
```
root@treehouses:~# treehouses memory
Your rpi has 946664 bytes of total memory with 745680 bytes used and 200004 bytes available
```

release-115 and remote 1830

sriharivishnu, 2019-12-21:
It was working for me:
[image]

release-111 and remote 1830

dun08015, 2019-12-21:
[image]
release-114 remote 1830

Nida8930, 2019-12-21:
[image]
Everything appears to be working for me. Treehouses 115, remote 1830, 3B+.

closed 2020-01-31 by dogi