treehouses/cli #559: add CHANGELOG.md

phamduchongan93, 2019-12-31:
Generate a record so that people can track bugs and upcoming features. This log should reflect the current features of the latest images. The downside is that we must have the members to monitor this file, making sure features are deliverable.

LordJashin32, 2020-01-01:
This sounds sort of ambiguous. If there is a bug that can't be squashed there should still be an issue open for it. Isn't github like a bug tracker at this point? If there is bugs that aren't on the issues page than yes I personally would like a list of them in a file somewhere. Features i think is covered by the help command. 

This changelog idea is great for customers/consumers. Idk but maybe make a change log here and then somewhere in planet or somewhere else they can pull it up?

dogi, 2020-01-02:
we add changelog automatically when putting it into https://npmjs.org

closed 2020-01-02 by dogi