treehouses/cli #642: treehouses tor list remove on space which is too much

dogi, 2020-01-30:
```
root@r120rpi4:/srv/kolibri# treehouses tor list
external <=> local
8081  <=> 8081
8080  <=> 8080
80  <=> 80
22  <=> 22
```
should be
```
root@r120rpi4:/srv/kolibri# treehouses tor list
external <=> local
8081 <=> 8081
8080 <=> 8080
80 <=> 80
22 <=> 22
```

jamjakpa, 2020-01-30:
> ```
> root@r120rpi4:/srv/kolibri# treehouses tor list
> external <=> local
> 8081  <=> 8081
> 8080  <=> 8080
> 80  <=> 80
> 22  <=> 22
> ```
> 
> should be
> 
> ```
> root@r120rpi4:/srv/kolibri# treehouses tor list
> external <=> local
> 8081 <=> 8081
> 8080 <=> 8080
> 80 <=> 80
> 22 <=> 22
> ```

@dogi , where do i find the file tor list , in module , I just see tor.sh  ,

jamjakpa, 2020-01-31:
> ```
> root@r120rpi4:/srv/kolibri# treehouses tor list
> external <=> local
> 8081  <=> 8081
> 8080  <=> 8080
> 80  <=> 80
> 22  <=> 22
> ```
> 
> should be
> 
> ```
> root@r120rpi4:/srv/kolibri# treehouses tor list
> external <=> local
> 8081 <=> 8081
> 8080 <=> 8080
> 80 <=> 80
> 22 <=> 22
> ```

@dogi , I change it here right, in that directory in that file: https://github.com/treehouses/cli/blob/master/modules/tor.sh