treehouses/cli #648: treehouses services <service> restart needs to always end with running service

dogi, 2020-02-01:
... when service is not running restart is not starting the service

kjong, 2020-02-01:
```
root@kjong:~/cli# ./cli.sh services planet stop
Stopping planet_planet_1  ... done
Stopping planet_couchdb_1 ... done
planet stopped
root@kjong:~/cli# ./cli.sh services running

root@kjong:~/cli# ./cli.sh services planet restart
planet stopped
Starting planet_couchdb_1 ... done
Starting planet_planet_1  ... done
Starting planet_db-init_1 ... done
planet built and started
tor active
80  <=> 80
```

kjong, 2020-02-01:
OK, it seems the issue is to do with services that are run using Docker and not Docker-compose (ie. nextcloud).

Nextcloud and portainer are being moved to compose soon so this should resolve itself soon.

kjong, 2020-02-06:
Nexcloud and Portainer were ported to docker-compose in #646.
This issue should be resolved.

closed 2020-02-06 by kjong