treehouses/cli #649: services: wrong ip address url

sriharivishnu, 2020-02-01:
```
root@sriharivishnu:~# treehouses services planet url both
169.254.4.234:80
169.254.4.234:2200 
...
```

kjong, 2020-02-01:
Will probably switch to using `ifconfig` instead of `hostname -I` to get the ip address.

vmnet8, 2020-02-06:
under release-121, it works well for me. both local and tor address are correct.
is it not a common problem?

kjong, 2020-02-06:
It wasn't a problem for everyone.

Basically, I was using `hostname -I` and taking the first result to get the ip address but the first result wasn't always the correct ip address.

So it would work for some but not for others.

The change to using `networkmode info` should produce more consistent results.

PR #660 (2020-02-06): treehouses services url with networkmode info (fixes #649)
Fixes #649

Changed from using `hostname` to using `treehouses networkmode` to get IP address when creating local url for services.
commit c3924257 (2020-02-06): treehouses services url with networkmode info (fixes #649) (#660)
closed 2020-02-06 by dogi