treehouses/cli #668: should use direct execution rather than calling bash

LordJashin32, 2020-02-07:
Right now we are using bash to execute scripts when we could just make the script executable, add a shebang (which it probably already has) and just execute directly. 
```bash $TEMPLATES/services/portainer/portainer_yml.sh```
->
```.$TEMPLATES/services/portainer/portainer_yml.sh```

kjong, 2020-02-08:
Removing the shebang from some scripts (per #666) and keeping them in others makes things very inconsistent and removes a lot of flexibility with regards to how we use the scripts (both executing directly and calling them in other scripts).

I'd rather have the shebang and not need it, than need it and not have it.

Anyways, as I mentioned in #666, we need the shebangs to ensure that the correct interpreter is used.

LordJashin32, 2020-02-13:
> Removing the shebang from some scripts (per #666) and keeping them in others makes things very inconsistent and removes a lot of flexibility with regards to how we use the scripts (both executing [...]

Wouldn't be removing the shebang this time. It would be instead of doing ```bash ./yml_create.sh```
we would do ```./yml_create.sh``` and just call it directly.

LordJashin32, 2020-02-13:
Just gonna close this my idea didn't work out. Owell. Better luck next time.

closed 2020-02-13 by LordJashin32