treehouses/cli #675: treehouses services <service_name> up connection error

Nida8930, 2020-02-10:
#### Overview

Periodically if the raspberry pi is unable to communicate with a service an error will appear and the process will stop. Error needs to be caught and either let the user know about the connection issue or automatically run the command again to successfully build and start the service.

#### Reproducing the issue

Running `treehouses services <service_name> up` can sometime display an error. 

I am attempting to get the message again so I can add a screenshot.

kjong, 2020-02-10:
Did this happen with the latest version of the cli?
Because the error handling that I added was merged very recently, and I'm thinking that that should catch the pull errors.

Nida8930, 2020-02-10:
@kjong This occurred on 1.13.4 so that very well could have been the issue. I am now updated to 1.13.13.