treehouses/cli #676: treehouses wificountry?

dogi, 2020-02-10:
```
$ treehouses wificountry
Success: the wifi country has been set to
```

is doing strange stuff
it should show which country it is set too

dogi, 2020-02-10:
is `treehouses default network` resetting wificountry?

jamjakpa, 2020-02-12:
i get on my pi Image version release-120 

[image]

Nida8930, 2020-02-12:
@jamjakpa this is an issue we have been having on the experimental image release [here](http://download.treehouses.io/experiment/treehouse-raspbiantwenty0.img.gz).

@dogi on the Rpi3B+ it says the wifi is unable to connect until wifi country is set using raspi-config. Raspi-config does not work until you run `treehouses default network` and reboot. After this you are able to set the wifi country using raspi-config and then use `treehouses wifi` as normal.

`treehouses default network` does reset the wifi country as far as I know, unless it is set somewhere besides wpa_supplicant.conf. Once the wifi country is initially set by raspi-config everything still works even if it is reset later by `default network`.

As far as the wificountry function goes, it sets the wifi country only. so `treehouses wificountry` with no argument should return an error or more needs to be modified to return what the current wifi country is set to if no argument is given. 

`treehouses wificountry <country>` appears to work as it should.

commit 7bb476b3 (2020-02-12): modified wificountry.sh (fixes #676)
commit 07a71270 (2020-02-13): treehouses wificountry (fixes #676) (#677)
closed 2020-02-13 by dogi