treehouses/cli #686: treehouses memory broken?

Evoong, 2020-02-13:
```
root@treehouses:~# treehouses memory
Your rpi has  bytes of total memory with  bytes used and  bytes available
root@treehouses:~# treehouses memory free

root@treehouses:~# treehouses memory total

root@treehouses:~#

```

Evoong, 2020-02-13:
Forgot this is my version numbers

```
root@treehouses:~# treehouses image
release-121
root@treehouses:~# treehouses version
1.13.16
```

dogi, 2020-02-13:
[image]

same error here ...

kjong, 2020-02-15:
```
root@kjong:~# treehouses memory
Your rpi has 1986000 bytes of total memory with 631664 bytes used and 1353732 bytes available
root@kjong:~# treehouses image
release-121
root@kjong:~# treehouses version
1.13.4
```

Was working with version 1.13.4.
After upgrading to the latest version 1.13.16, `treehouses memory` no longer works.

PR #687 (2020-02-15): treehouses memory with global vars (fixes #686)
Fixes #686

Since the main `memory()` function calls the other functions, we can't use local variables since local variables will be out of scope.

Reverting back to global variables fixes the issues.
commit 25f96db9 (2020-02-15): treehouses memory with global vars (fixes #686) (#687)
closed 2020-02-15 by dogi