treehouses/cli #706: treehouses led - dance broken

Neethu-Mohan, 2020-02-20:
Error: led dance not found

dogi, 2020-02-20:
```
pi@pinode1:~/cli $ sudo ./cli.sh led onam                                                                                                        
leds are set to onam mode.                                                                                                                       
Look at your Rpi leds, both leds will be in this pattern...
Green LED: 5 blink
Both LED: 1 sec off
Red LED: 5 blink
Green LED: 5 blink
Both LED: 1 sec
Red LED: 5 blink
Error: led 'onam' is not present 
```

dogi, 2020-02-20:
https://github.com/treehouses/cli/blob/master/modules/led.sh#L2

dogi, 2020-02-20:
and

```
-rwxrwxr-x 1 nemo nemo 11456 Feb 20 00:29 led.sh
```

remove executable flag

PR #707 (2020-02-20): treehouses led no locals (fixes #706)
Fixes #706
commit 3d24f192 (2020-02-20): treehouses led no locals (fixes #706) (#707)
closed 2020-02-20 by dogi