treehouses/cli #722: treehouses detectrpi outdated

LordJashin32, 2020-02-23:
From the raspberrypi website:
```
Why does cpuinfo report I have a BCM2835?
The upstream Linux kernel developers had decided that all models of Raspberry Pi return bcm2835 as the SoC name. At Raspberry Pi we like to use as much upstream kernel code as possible, as it makes software maintenance much easier, so we use this code. Unfortunately it means that cat /proc/cpuinfo is inaccurate for the Raspberry Pi 2, Raspberry Pi 3 and Raspberry Pi 4, which use the bcm2836/bcm2837, bcm2837 and bcm2711 respectively. You can use cat /proc/device-tree/model to get an accurate description of the SoC on your Raspberry Pi model.
```
So we should be using ```/proc/device-tree/model``` instead of ```/proc/cpuinfo``` where possible.

dogi, 2020-02-25:
https://www.cnx-software.com/2020/02/24/raspberry-pi-4-rev-1-2-fixes-usb-c-power-issues-improves-sd-card-resilience/

dogi, 2020-02-27:
`cat /sys/firmware/devicetree/base/model`

commit 075914cb (2020-02-27): changed detectrpi.sh file (fixes #722)
commit 3aee27a2 (2020-02-29): treehouses detectrpi full (fixes #722) (#743)
closed 2020-02-29 by dogi