treehouses/cli #760: investigation into RAM usage w/services

LordJashin32, 2020-03-02:


kjong, 2020-03-07:
We can use `top`, which is already installed.
Or we can also use `netdata`.

If we want to have automatic alerts, there is https://github.com/XavierBerger/RPi-Monitor which looks like it has alert functionality.

```
Processing metrics and raising alerts
RPi-Monitor embed an alert engine fully configurable and able to determine if metrics are out of expected range and raise alerts.

When an alert is detected, RPi-Monitor can execute an action. For example, sending email or SMS.

Definition of critera of alert detection as well as action to perform when alert is detected are fully configurable.
```

LordJashin32, 2020-03-08:
> We can use `top`, which is already installed.
> Or we can also use `netdata`.
> 
> If we want to have automatic alerts, there is [XavierBerger/RPi-Monitor](https://github.com/XavierBerger/RPi-Monitor) which looks like it has alert functionality.
> 
> ```
> Processing metrics and raising alerts
> RPi-Monitor embed an alert engine fully configurable and able to determine if metrics are out of expected range and raise alerts.
> 
> When an alert is detected, RPi-Monitor can execute an action. For example, sending email or SMS.
> 
> Definition of critera of alert detection as well as action to perform when alert is detected are fully configurable.
> ```
> 
> But I don't really like the idea of having so many similar programs installed.

I was thinking of making sure a service can't be turned on if we know it's going to hit the memory limit of the Pi with the services currently running + the new service coming online. But this is interesting to read. Idk what solution is best or if we need to or not. I think eventually we might.