treehouses/cli #846: treehouses services <name> <icon> ??

sriharivishnu, 2020-03-14:
Add a command to get the logo of a service

LordJashin32, 2020-03-14:
One possible way to do it: https://github.com/tootsuite.png
it only works for organizations though. So if the creator doesn't have the same image. Then we have to go outside github for them. Or just have a default.

kjong, 2020-03-16:
Are you talking about getting a local image or getting one from online?

dogi, 2020-03-19:
@kjong and @LordJashin32 think that SVGs as part of our services install files would be awesome

dogi, 2020-03-19:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   id="svg2"
   version="1.1"
   viewBox="0 0 400 400"
   height="500"
   width="500">
  <metadata
     id="metadata28">
    <rdf:RDF>
      <cc:Work
         rdf:about="">
        <dc:format>image/svg+xml</dc:format>
        <dc:type
           rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
        <dc:title></dc:title>
      </cc:Work>
    </rdf:RDF>
  </metadata>
  <defs
     id="defs26" />
  <g
     transform="scale(3.8461538,3.8461538)"
     id="g4171">
    <circle
       style="fill:#ffffff;fill-opacity:1;stroke:none;stroke-width:1.5;stroke-linejoin:round;stroke-miterlimit:4;stroke-dasharray:none;stroke-opacity:0.00392157"
       id="path4169"
       cx="52"
       cy="52"
       r="52" />
    <g
       id="g4158">
      <path
         id="path6"
         d="M 38.22,5.29 C 44.53,3.22 51.78,8.34 51.9,14.98 52.38,20.97 47.04,26.67 40.97,26.3 34.75,26.34 29.42,20.12 30.65,13.96 31.25,9.95 34.29,6.38 38.22,5.29 Z"
         style="fill:#327bb9" />
      <path
         id="path8"
         d="m 64.4,6.54 c 1,-1.14 3.09,-1 3.5,0.63 1.88,4.61 0.15,9.72 -1.64,14.09 C 61.62,32.22 49.84,39.6 37.94,38.85 29.14,38.44 20.4,33.95 15.43,26.62 c -1.08,-1.65 -2.55,-5.58 0.55,-5.84 4.12,0.47 7.25,3.7 10.98,5.3 9.32,5.08 21.93,2.73 28.82,-5.32 3.86,-4.12 5.18,-9.86 8.62,-14.22 z"
         style="fill:#327bb9" />
      <path
         id="path10"
         d="m 81.38,15.45 c 1.74,-1.1 4.5,0.14 3.66,2.43 -1.33,4.3 -4.52,7.73 -6.1,11.94 -4.42,9.97 -0.62,22.74 8.48,28.72 3.59,2.62 8.01,3.8 11.54,6.49 1.4,0.86 1.57,3.2 -0.15,3.81 C 95.15,70.44 91,69.46 87.36,68.31 78.01,65.34 70.48,57.4 67.98,47.92 64.66,35.91 69.85,21 81.38,15.45 Z"
         style="fill:#327bb9" />
      <path
         id="path12"
         d="m 86.26,32.24 c 5.9,-2.6 13.36,1.41 14.52,7.72 1.29,5.72 -2.88,12.02 -8.74,12.8 -5.99,1.18 -12.23,-3.73 -12.44,-9.83 -0.33,-4.5 2.44,-9.06 6.66,-10.69 z"
         style="fill:#327bb9" />
      <path
         id="path14"
         d="m 7.35,37.53 c 12.52,-1.67 24.85,7.58 28.9,19.22 3.25,9.07 1.66,19.69 -4.15,27.39 -2.42,3.15 -5.28,6.4 -9.26,7.47 -1.16,0.44 -3.04,0 -3.03,-1.52 -0.13,-2.21 1.33,-4.07 2.31,-5.92 2.71,-4.34 5.23,-9.03 5.63,-14.23 C 28.4,61.42 24.34,52.56 17.02,47.99 13.41,45.42 8.88,44.35 5.54,41.41 3.81,39.94 5.49,37.64 7.35,37.53 Z"
         style="fill:#327bb9" />
      <path
         id="path16"
         d="m 49.68,41.78 c 0.91,-0.07 2.74,-0.19 3.65,-0.26 0.18,2.81 0.38,5.61 0.54,8.41 2.35,-1.53 4.73,-3.02 7.07,-4.56 0.51,0.77 1.51,2.31 2.02,3.07 -2.35,1.51 -4.67,3.05 -7.02,4.56 2.49,1.25 4.96,2.53 7.46,3.78 -0.55,1.09 -1.1,2.18 -1.65,3.27 -2.49,-1.26 -4.99,-2.48 -7.48,-3.75 0.18,2.78 0.34,5.57 0.51,8.35 -1.22,0.08 -2.43,0.16 -3.65,0.24 -0.18,-2.81 -0.38,-5.62 -0.56,-8.42 -2.35,1.55 -4.73,3.05 -7.1,4.57 -0.66,-1.03 -1.31,-2.06 -1.96,-3.09 2.33,-1.47 4.7,-2.91 6.81,-4.69 -2.44,-1.17 -4.85,-2.41 -7.26,-3.63 0.55,-1.09 1.1,-2.18 1.64,-3.27 2.48,1.26 4.98,2.48 7.47,3.74 C 50,47.33 49.85,44.55 49.68,41.78 Z"
         style="fill:#87b852" />
      <path
         id="path18"
         d="M 11.35,54.43 C 18.81,51.68 27.22,59.29 25.05,67 23.69,74.77 13.09,78.14 7.47,72.63 1.29,67.62 3.71,56.51 11.35,54.43 Z"
         style="fill:#327bb9" />
      <path
         id="path20"
         d="m 59.43,68.59 c 8.43,-1.8 17.6,0.27 24.36,5.64 3.14,2.47 6.35,5.53 7.04,9.66 0.25,2 -2.1,2.66 -3.58,1.91 C 83.16,84.12 79.69,81.21 75.52,79.68 66.55,76 55.39,78.76 49.1,86.11 c -3.44,3.86 -5.11,8.83 -7.75,13.18 -0.74,1.59 -3.42,2.68 -4.25,0.58 C 35.29,96 36.49,91.54 37.78,87.7 41.07,78.16 49.52,70.61 59.43,68.59 Z"
         style="fill:#327bb9" />
      <path
         id="path22"
         d="m 59.34,81.43 c 6.25,-3.04 14.46,1.65 14.99,8.57 0.75,5.62 -3.62,11.3 -9.29,11.85 -6.23,1.06 -12.48,-4.57 -12.03,-10.89 0.07,-4.03 2.62,-7.93 6.33,-9.53 z"
         style="fill:#327bb9" />
    </g>
  </g>
</svg>
```

dogi, 2020-03-19:
or new attempt
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg xmlns="http://www.w3.org/2000/svg" xml:space="preserve" width="8.26772in" height="11.6929in" version="1.1" style="shape-rendering:geometricPrecision; text-rendering:geometricPrecision; image-rendering:optimizeQuality; fill-rule:evenodd; clip-rule:evenodd"
viewBox="0 0 8268 11693"
 xmlns:xlink="http://www.w3.org/1999/xlink">
 <defs>
  <style type="text/css">
   <![CDATA[
    .fil0 {fill:#0070B0}
    .fil1 {fill:#60AE36}
   ]]>
  </style>
 </defs>
 <g id="Layer_x0020_1">
  <metadata id="CorelCorpID_0Corel-Layer"/>
  <g id="_2195878499632">
   <path class="fil0" d="M5353 4672c0,882 331,1352 830,1790 283,249 1372,906 1622,844 45,-167 -152,-207 -275,-297 -1307,-957 -1871,-2490 -1354,-4057 36,-111 184,-387 94,-443 -98,-60 -232,223 -282,307 -277,476 -635,1169 -635,1856z"/>
   <path class="fil0" d="M461 4373c-49,182 140,169 468,441 540,446 939,973 1174,1681 562,1697 -467,2944 -25,2666 42,-27 1343,-1777 598,-3213 -228,-441 -535,-719 -917,-979 -164,-112 -1092,-648 -1298,-596z"/>
   <path class="fil0" d="M2667 9511c182,49 169,-140 440,-468 447,-540 974,-939 1681,-1173 1698,-563 2945,466 2667,25 -27,-43 -1777,-1343 -3213,-599 -441,228 -720,536 -979,917 -112,165 -648,1092 -596,1298z"/>
   <path class="fil0" d="M2472 3984c-504,0 -881,-60 -1342,-228 -138,-50 -231,-133 -328,-53 -94,153 1340,916 2163,917 883,0 1353,-332 1791,-831 248,-283 906,-1372 843,-1622 -194,-52 -240,282 -571,623 -265,273 -339,358 -666,581 -476,324 -1158,613 -1890,613z"/>
   <circle class="fil0" cx="7089" cy="4879" r="760"/>
   <circle class="fil0" cx="3179" cy="2904" r="760"/>
   <circle class="fil0" cx="1179" cy="6789" r="760"/>
   <circle class="fil0" cx="5102" cy="8789" r="760"/>
   <polygon class="fil1" points="4981,5332 5105,5717 4122,5932 3265,6303 3140,5918 4068,5721 "/>
   <polygon class="fil1" points="4972,6315 4701,6621 4023,5877 3273,5320 3544,5014 4179,5719 "/>
   <polygon class="fil1" points="4127,4836 4531,4912 4226,5871 4118,6799 3714,6723 4007,5821 "/>
  </g>
 </g>
</svg>
```

kjong, 2020-03-19:
Do we want to include a .svg file or add the svg code to the install scripts?

Do we want to return a image, the .svg file, or the svg code to remote?

PR #882 (2020-03-19): treehouses services <service> icon (fixes #846)
Fixes #846
commit 5a8b5f9a (2020-03-19): treehouses services <service> icon (fixes #846) (#882)
closed 2020-03-19 by dogi