treehouses/cli #866: tests tor

kjong, 2020-03-16:
https://github.com/treehouses/cli/blob/838ee2a8b2d89992b8be1a1d5a90a89c18189921/tests/tor.bats#L54-L62

The last tests being run on tor are `tor deleteall` and `tor destroy`.
Are we assuming that the tests are only being run on a clean system?

LordJashin32, 2020-03-16:
> https://github.com/treehouses/cli/blob/838ee2a8b2d89992b8be1a1d5a90a89c18189921/tests/tor.bats#L54-L62
> 
> The last tests being run on tor are `tor deleteall` and `tor destroy`.
> Are we assuming that the tests are only being run on a clean system (not being used before / after the tests)?

For you, I would avoid running all the tests at once. Because we haven't worked out if it works on ssh/wifi/bluetooth only yet. We are working on it though. Right now it has some problems on reboot. 

We want it to be able to work no matter what. After clean install/after using all tests/ etc.

LordJashin32, 2020-03-29:
I wasn't sure what our default setup was. I guess we are running Tor by default? So I'll mix the commands around so Tor is running instead of being destroyed. This should ideally work with a default image and keep it mostly the same. Trying to make everything return back to how it was before isn't possible. This is mainly for testing for developers. Not users. This is meant for us so when we have huge changes to Raspbian, our code, our dependencies, or whatever. We can run the tests and see what is failing and what is good.