treehouses/cli #894: treehouses url local

kjong, 2020-03-21:
```
root@kjong:~# treehouses services planet url
:80
:2200
masrx3fngdcup6cwihvplcgae5c6soppks425jplux24snv3nkfwtxad.onion:80
masrx3fngdcup6cwihvplcgae5c6soppks425jplux24snv3nkfwtxad.onion:2200

root@kjong:~# treehouses services pihole url
:8053/admin
masrx3fngdcup6cwihvplcgae5c6soppks425jplux24snv3nkfwtxad.onion:8053/admin
```

ip address not being appended to local url

dogi, 2020-03-21:
```
$ treehouses upgrade 1.15.12
$ treehouses services planet url local
192.168.68.110:80
192.168.68.110:2200
```

kjong, 2020-03-21:
```
root@kjong:~/cli# ./cli.sh services planet url both
:80
:2200
qcaz7y6ftq2v7s53ba75kh6svn5uqbc3tiu7tipwqgcwvrinumsxviyd.onion:80
qcaz7y6ftq2v7s53ba75kh6svn5uqbc3tiu7tipwqgcwvrinumsxviyd.onion:2200
root@kjong:~/cli# ./cli.sh services planet url
192.168.1.154:80
192.168.1.154:2200
qcaz7y6ftq2v7s53ba75kh6svn5uqbc3tiu7tipwqgcwvrinumsxviyd.onion:80
qcaz7y6ftq2v7s53ba75kh6svn5uqbc3tiu7tipwqgcwvrinumsxviyd.onion:2200
```

kjong, 2020-03-21:
Changing `checkargn 2` in `modules/networkmode.sh` to `checkargn 3` seems to fix it.

kjong, 2020-03-23:
Fixed in #898

closed 2020-03-23 by kjong