treehouses/cli #923: treehouses services available adsfkhsdj

dogi, 2020-03-26:
```                                                                                                                                   
root@r125rpi4:~# treehouses services available
couchdb
kolibri
mariadb
mastodon
moodle
nextcloud
pihole
planet
root@r125rpi4:~# treehouses services available all
couchdb
kolibri
mariadb
mastodon
moodle
nextcloud
pihole
planet
root@r125rpi4:~# treehouses services available allfklgjsadgkjsd
couchdb
kolibri
mariadb
mastodon
moodle
nextcloud
pihole
planet
```
please check for too much input
treehouses services available

kjong, 2020-03-26:
This is supposed to be handled by `checkargn`

```
root@kjong:~# treehouses services available asd asd asd
Error: Too many arguments.
Available Services:
planet       Planet Learning is a generic learning system built in Angular & CouchDB
kolibri      Kolibri is a learning platform using DJango
nextcloud    Nextcloud is a safe home for all your data, files, etc
netdata      Netdata is a distributed, real-time performance and health monitoring for systems
```

Right now `checkargn` is at the very top so we can't control how many arguments to check for for each specific function.

Quick fix would be to remove `checkargn` from the top and instead add it to each function with the number of arguments being checked changed for each specific function.

Would this approach work?

LordJashin32, 2020-03-26:
> This is supposed to be handled by `checkargn`
>
> ```
> root@kjong:~# treehouses services available asd asd asd
> Error: Too many arguments.
> Available Services:
> planet       Planet Learning is a generic [...]

Yes i think this is fine. But i would leave the checkargn at the top as well. Why not. You can do it either way though. Probably the less code the better for that.

PR #925 (2020-03-26): treehouses services checkargn (fixes #923)
Fixes #923

Added `checkargn` to each "function" in order to more accurately check for number of arguments.
commit 24176a2d (2020-03-26): treehouses services checkargn (fixes #923) (#925)
closed 2020-03-26 by dogi