treehouses/cli #932: treehouses services service not found

LordJashin32, 2020-03-26:
This is on fresh install and here's the proof its fresh install. And idk if this is intended output or not but just throwing it in here for now. 
```
root@treehouses:~# treehouses image
release-125
root@treehouses:~# treehouses version
1.14.24
root@treehouses:~# treehouses upgrade

> @treehouses/cli@1.14.24 postuninstall /usr/lib/node_modules/@treehouses/cli
> if [ $(id -u) = 0 ]; then rm /etc/bash_completion.d/_treehouses; if [ -f /etc/treehouses.conf ]; then rm /etc/treehouses.conf; fi fi && exit 0

/usr/bin/treehouses -> /usr/lib/node_modules/@treehouses/cli/cli.sh

> @treehouses/cli@1.16.21 postinstall /usr/lib/node_modules/@treehouses/cli
> if [ $(id -u) = 0 ]; then ln -sr _treehouses /etc/bash_completion.d/_treehouses; fi && exit 0

+ @treehouses/cli@1.16.21
updated 1 package in 3.524s
root@treehouses:~# treehouses version
1.16.21
root@treehouses:~# treehouses image
release-125
root@treehouses:~# treehouses services available
couchdb
kolibri
mariadb
mastodon
moodle
netdata
nextcloud
ntopng
pihole
planet
portainer
privatebin
seafile
root@treehouses:~# treehouses services kolibri install
Pulling kolibri ... done
kolibri installed
root@treehouses:~# treehouses services kolibri start
kolibri not found
root@treehouses:~# treehouses services kolibri stop
kolibri not found
root@treehouses:~# treehouses services kolibri up
Creating network "kolibri_default" with the default driver
Creating kolibri_kolibri_1 ... done
kolibri built and started
root@treehouses:~# treehouses services kolibri start
Starting kolibri ... done
kolibri started
root@treehouses:~# treehouses services kolibri stop
Stopping kolibri_kolibri_1 ... done
kolibri stopped
root@treehouses:~# treehouses services kolibri down
Removing kolibri_kolibri_1 ... done
Removing network kolibri_default
kolibri stopped and removed
root@treehouses:~# treehouses services kolibri stop
kolibri not found
root@treehouses:~# treehouses services kolibri start
kolibri not found
root@treehouses:~# treehouses services kolibri up
Creating network "kolibri_default" with the default driver
Creating kolibri_kolibri_1 ... done
kolibri built and started
root@treehouses:~# treehouses services kolibri stop
Stopping kolibri_kolibri_1 ... done
kolibri stopped
root@treehouses:~#

```

dogi, 2020-03-26:
@LordJashin32 nice catch we should maybe rethink our outputs

kjong, 2020-03-26:
https://github.com/treehouses/cli/pull/927#issuecomment-604743668

PR #940 (2020-03-28): treehouses services improve outputs (fixes #932)
Fixes #932

Added suggestions to try `up` first before trying `start` or `stop`.
Added usage examples for service specific commands.
commit 4fa37e57 (2020-03-28): treehouses services improve outputs (fixes #932) (#940)
closed 2020-03-28 by dogi