treehouses/cli #942: treehouses services environment variables

kjong, 2020-03-28:
Configuring environment variables for services

`treehouses services <service> environment` :
```
if [ $(source $SERVICES/install-${service_name}.sh && uses_env) = true ]; then
  if [ -e /srv/$service_name/.env ]; then
    vim /srv/$service_name/.env
  else
    echo "ERROR: /srv/$service_name/.env not found"
    echo "try running '$BASENAME services $service_name install' first"
    exit 1
  fi
else
  echo "$service_name does not take environment variables"
fi
```

/srv/$service_name/.env :
```
  NAME_VAR=name
  ...
```

/srv/$service_name/$service_name.yml :
```
  environment:
    - NAME=${NAME_VAR}
    ...
```

/services/install-${service_name}.sh :
```
function uses_env {
  return true / false
}
```
```
function install {
  # create .env
  {
    echo "NAME_VAR="
  } > /srv/$service_name/.env
```

### Potential Issues
* What happens if .env has empty environment variables (eg. `- NAME=`) when doing `up`? Need to ensure that .env file is properly configured before trying to create container.
* Possible to edit .env files using vim via Remote?

LordJashin32, 2020-03-29:
Need to also be careful of variable name conflicts between services. If they both use the same variable name we could run into trouble. I assume these would be global variables. Anyway. Its a cool idea.

kjong, 2020-03-30:
I was thinking of using separate environment files for each service. While it would be "easier" to use a single file, it would not be very fun trying to make sure there are no variable conflicts like you said.

Still fleshing out this feature so more suggestions are welcome.

PR #1037 (2020-04-25): treehouses services <service> environment (fixes #942)
Fixes #942

`treehouses services <service> environment edit`:
opens vim so user can edit .env

`treehouses services <service> environment check`:
outputs the docker-compose .yml with values from .env

#### TODO
- [x] add `--project-directory` to all docker commands
- [x] add `environment edit` to `install` ~and / or `up`~
- [x] find a way to grep (?) for warnings (`validate_yml`) / grep .env instead (/n, space, tab after =)
- [x] update install scripts
- [x] update help, readme, ...etc.
- [x] update services/README.md
- [x] commented out e [... cut]
commit 37b6c9ec (2020-04-25): treehouses services <service> environment (fixes #942) (#1037)
closed 2020-04-25 by dogi