treehouses/cli #1013: add treehouses remote services status

sriharivishnu, 2020-04-04:
A command to return all of the services' status (including running, installed, available, icon, info, autorun data).
If it is possible, perhaps the best way would be in JSON format as a string or perhaps each section of the string is labeled with a unique identifier to separate sections/ services.

kjong, 2020-04-04:
```
{
  "available":["couchdb", "kolibri", "mariadb", "mastodon"],
  "installed":["couchdb", "kolibri", "mastodon"],
  "running":["couchdb", "mastodon"],
  "icon":{
      "couchdb":"<svg>",
      "kolibri":"<svg>",
      "mariadb":"<svg>",
      "mastodon":"<svg>"
   }
  "info":{
      "couchdb":"<info>",
      "kolibri":"<info>",
      "mariadb":"<info>",
      "mastodon":"<info>"
   }
  "autorun":{
      "couchdb":"true/false",
      "kolibri":"true/false",
      "mariadb":"true/false",
      "mastodon":"true/false"
   }
}
```

Would a json like this work?

sriharivishnu, 2020-04-04:
The JSON object itself as a string worked, however, the second code string (one-liner) did not. One mistake I found was the `\/` character, which was not recognized.

kjong, 2020-04-04:
I'm assuming there's some character in there that needs to be escaped. Once I have the final output I'll show it to you so that we can check again.

sriharivishnu, 2020-04-04:
Ok thanks!

PR #1021 (2020-04-06): `treehouses remote json` (fixes #1013)
Fixes #1013

@sriharivishnu
I'm not sure how you process it on your end. Let me know if there any changes that need to be made regarding escapes or quotations and such.
commit d94bd984 (2020-04-06): `treehouses remote json` (fixes #1013) (#1021)
closed 2020-04-06 by dogi