treehouses/cli #1137: `treehouses usb on/off` broke if device connected?

LordJashin32, 2020-04-28:
```bash
root@treehouses:~/cli# hub-ctrl
Hub #0 at 002:001
 INFO: ganged switching.
 WARN: Port indicators are NOT supported.
Hub #1 at 001:002
 INFO: ganged switching.
Hub #2 at 001:001
 INFO: individual power switching.
 WARN: Port indicators are NOT supported.
root@treehouses:~/cli# treehouses usb on
failed to control.
: Broken pipe
failed to control.
: Broken pipe
failed to control.
: Broken pipe
usb ports turned on
root@treehouses:~/cli# treehouses usb off
failed to control.
: Broken pipe
failed to control.
: Broken pipe
failed to control.
: Broken pipe
failed to control.
: Protocol error
usb ports turned off
root@treehouses:~/cli#

```

kjong, 2020-05-11:
I'm not sure if there's anything we can do about this since we're using a third party package (that hasn't been updated in 6 years).

Maybe we should just deprecate `usb.sh` since it wasn't that great even from the beginning.

LordJashin32, 2020-05-15:
yeah haven't researched much into this one yet. It'd be cool to find a solution but just like other hardware specific packages. might not be any.