treehouses/cli #1241: `treehouses remote help json`

sriharivishnu, 2020-05-09:
Output help menu as a json

("command": "description")

kjong, 2020-05-09:
```
{
  "bluetooth": {
    "on": "This will start the bluetooth server, which lets the user control the raspberry pi using the mobile app.",
    "off": "This will stop the bluetooth server, and bring everything back to regular mode."
  }
}
```

We could do something like this but not sure how we can include a description for the top level command (in this case, the description for bluetooth itself).

```
{
  "bluetooth": {
    "bluetooth": "Switches between hotspot / regular bluetooth mode, or displays the bluetooth mac address",
    "on": "This will start the bluetooth server, which lets the user control the raspberry pi using the mobile app.",
    "off": "This will stop the bluetooth server, and bring everything back to regular mode."
  }
}
```

Maybe something like this to include the top level command description?

sriharivishnu, 2020-05-09:
Honestly, the whole output is good.
`treehouses help bluetooth`
```
Switches between hotspot / regular bluetooth mode, or displays the bluetooth mac address

Example:
  treehouses bluetooth
      on

  treehouses bluetooth status
      bluetooth service status: on
      rpibluetooth service status: on

  treehouses bluetooth on
      This will start the bluetooth server, which lets the user control the raspberry pi using the mobile app.

  treehouses bluetooth off
      This will stop the bluetooth server, and bring everything back to regular mode.
      This will also remove the bluetooth device id.

  treehouses bluetooth pause
      Performs the same as 'treehouses bluetooth off'
      The only difference is that this command will not remove the bluetooth device id.

  treehouses bluetooth restart
      This will restart the bluetooth server using treehouses bleutooth 'off' and 'on'

  treehouses bluetooth mac
      This will display the bluetooth MAC address

  treehouses bluetooth id
      This will display the network name along with the bluetooth id number

  treehouses bluetooth button
      When the GPIO pin 18 is on the bluetooth will ne turned off
      Otherwise the bluetooth mode will be changed to hotspot

  treehouses bluetooth id number
      This will display the bluetooth id number

  treehouses bluetooth log
      This will display the logs of bluetooth services

  treehouses bluetooth log follow
      This will display the logs as they come in live of the bluetooth services
      press (ctrl + c) to exit

```

kjong, 2020-05-09:
Do you just mean something like
```
{
  "bluetooth":"output of `treehouses help bluetooth`"
}
```

PR #1249 (2020-05-09): treehouses remote help json (fixes #1241)
Fixes #1241

Takes a bit under 3 seconds on my machine.

Line breaks are included in the descriptions.
commit f2030c49 (2020-05-09): treehouses remote help json (fixes #1241) (#1249)
closed 2020-05-09 by dogi