treehouses/cli #1272: treehouses help hard to read

jiaxuan-2, 2020-05-15:
[image]

LordJashin32, 2020-05-15:
maybe we could check for terminal width/height. and adjust helped based on that. as a sort of responsiveness. It would be better to multiline some of these if they don't fit. Maybe there's some solution to prettify this output based on not allowing it to carryon to new line. like word wrap?

LordJashin32, 2020-05-15:
Here's our options. 
We can pipe the output of treehouses help to `less -S` to allow horizontal/vertical scrolling.
Or we can use  `setterm -linewrap on` ` setterm -linewrap off`
but no horizontal scrolling.
Or we can provide both options.

kjong, 2020-05-16:
I think the problem here is more that there are lines that are too long and should be broken up into 2 or more lines.

Look at `ap` for instance. It's 135 columns long.
I feel like the lines should be shorter than 50 columns or so.

LordJashin32, 2020-05-16:
agreed