treehouses/cli #1295: should we care about this error or nah? services

LordJashin32, 2020-05-18:
```bash
root@treehouses:~/cli# ./cli.sh services couchdb cleanup
service autorun set to false
Removing network couchdb_default
WARNING: Network couchdb_default not found.
Removing image treehouses/couchdb:2.3.1
ERROR: Failed to remove image for service couchdb: 409 Client Error: Conflict ("b'conflict: unable to remove repository reference "treehouses/couchdb:2.3.1" (must force) - container ce532e317f58 is using its referenced image e6a38d57a53c'")
couchdb stopped and removed
couchdb cleaned up
root@treehouses:~/cli#
```

kjong, 2020-05-21:
This is most likely happening because couchdb is also used by planet, which we don't want to remove. So the error can be ignored.

Maybe we could skip removing the image during cleanup.

dogi, 2020-05-23:
or we check if use by other services