treehouses/cli #1315: can we detect if there is a camera plugged in

dogi, 2020-05-21:


LordJashin32, 2020-05-23:
```bash
v4l2-ctl --list-device
root@treehouses:/home/pi/Videos# v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12

mmal service 16.1 (platform:bcm2835-v4l2):
	/dev/video0

root@treehouses:/home/pi/Videos#
```

Camera Version | Max (microseconds)
-- | --
V1 (OV5647) | 6000000 (i.e. 6s)
V2 (IMX219) | 10000000 (i.e. 10s)
HQ (IMX477) | 200000000 (i.e. 200s)

LordJashin32, 2020-05-23:
If we want to detect which camera version looks like we will have to check max resolutions. And if we want to detect if camera is connected or not we just have to try and take a picture and if it returns the right error message then there's no camera still connected.

LordJashin32, 2020-05-23:
We can use the values here https://www.raspberrypi.org/documentation/hardware/camera/

PR #1353 (2020-05-29): `treehouses camera detect` (fixes #1315)
allows the user to detect if there is a camera plugged in, and what version of the camera it is.

- needs testing for HQ camera and camera v2
- IR camera is v2
- nothing added to test.bats

fixes #1315
commit 070a80b0 (2020-06-01): `treehouses camera detect` (fixes #1315) (#1353)
closed 2020-05-29 by dogi