treehouses/cli #1328: `bats tests/magazine.bats` fails

dogi, 2020-05-24:
... because latest already exists
```
root@treehouses:~/cli# bats tests/magazine.bats
 ✗  magazine magpi
   (from function `assert_success' in file /usr/lib/node_modules/bats-assert/src/assert.bash, line 114,
    in test file tests/magazine.bats, line 6)
     `assert_success' failed

   -- command failed --
   status : 1
   output : MagPi93.pdf already exists, exiting...
   --

 ✓  magazine magpi latest
 ✓  magazine magpi all

3 tests, 1 failure
```

jayala-29, 2020-05-27:
this test assumes there are no existing magpi magazines on the user's device, maybe a check inside the test to remove existing ones?

jayala-29, 2020-05-28:
I will fix this once the other magpi merges are made :)

PR #1371 (2020-05-30): `treehouses magazine magpi` tests (fixes #1328)
fixes #1328

- updated tests in the bats file
commit a93a8b88 (2020-05-30): `treehouses magazine magpi` tests (fixes #1328) (#1371)
closed 2020-05-30 by dogi