treehouses/cli #1377: `treehouses upgrade bluetooth` returns an incorrect message in Vagrant

law-lin, 2020-05-30:
`treehouses upgrade bluetooth` may need a checkrpi so it doesn't run in Vagrant.
```
root@cli:~# treehouses upgrade bluetooth
cp: cannot stat '/usr/local/bin/bluetooth-server.py': No such file or directory
^[Successfully updated and restarted bluetooth server
```

dogi, 2020-06-02:
it should not work in vagrant

warthog16, 2020-06-03:
```
root@cli:/home/vagrant/cli/modules# treehouses version
1.20.35
root@cli:/home/vagrant/cli/modules# treehouses upgrade

> @treehouses/cli@1.20.35 postuninstall /usr/lib/node_modules/@treehouses/cli
> if [ $(id -u) = 0 ]; then rm /etc/bash_completion.d/_treehouses; if [ -f /etc/treehouses.conf ]; then rm /etc/treehouses.conf; fi fi && exit 0

/usr/bin/treehouses -> /usr/lib/node_modules/@treehouses/cli/cli.sh

> @treehouses/cli@1.21.25 postinstall /usr/lib/node_modules/@treehouses/cli
> if [ $(id -u) = 0 ]; then ln -srf _treehouses /etc/bash_completion.d/_treehouses; fi && exit 0

+ @treehouses/cli@1.21.25
updated 1 package in 2.956s

   ╭────────────────────────────────────────────────────────────────╮
   │                                                                │
   │      New patch version of npm available! 6.14.4 → 6.14.5       │
   │   Changelog: https://github.com/npm/cli/releases/tag/v6.14.5   │
   │               Run npm install -g npm to update!                │
   │                                                                │
   ╰────────────────────────────────────────────────────────────────╯

root@cli:/home/vagrant/cli/modules# treehouses upgrade bluetooth
Error: Must be run with rpi system
```

closed 2020-06-03 by warthog16