treehouses/cli #1392: fix `bats tests/magazine.bats`

dogi, 2020-05-30:
@law-lin wrote:

```
> Everything seems to work, all the PDF files are installed correctly. Are these BATS tests supposed to fail though?
> 
> ```
> root@treehouses:~/cli# bats tests/magazine.bats
>  ✓  magazine magpi
>  ✓  magazine magpi latest
>  ✓  magazine magpi all
>  ✓  magazine hackspace
>  ✗  magazine hackspace latest
>    (from function `assert_success' in file /usr/lib/node_modules/bats-assert/src/assert.bash, line 114,
>     in test file tests/magazine.bats, line 26)
>      `assert_success' failed
> 
>    -- command failed --
>    status : 1
>    output : HackSpace31.pdf already exists, exiting...
>    --
> 
>  ✓  magazine hackspace all
>  ✓  magazine wireframe
>  ✗  magazine wireframe latest
>    (from function `assert_success' in file /usr/lib/node_modules/bats-assert/src/assert.bash, line 114,
>     in test file tests/magazine.bats, line 41)
>      `assert_success' failed
> 
>    -- command failed --
>    status : 1
>    output (12 lines):
>      --2020-05-30 05:21:53--  [https://wireframe.raspberrypi.org/issues](https://wireframe.raspberrypi.org/issues?q=is%3Aissue+is%3Aopen+author%3Adogi+archived%3Afalse+sort%3Aupdated-desc)
>      Resolving wireframe.raspberrypi.org (wireframe.raspberrypi.org)... 104.22.0.43, 104.22.1.43, 2606:4700:10::6816:12b, ...
>      Connecting to wireframe.raspberrypi.org (wireframe.raspberrypi.org)|104.22.0.43|:443... connected.
>      HTTP request sent, awaiting response... 200 OK
>      Length: unspecified [text/html]
>      Saving to: ‘issues’
> 
>           0K .......... .......... .......... ....                   352K=0.1s
> 
>      2020-05-30 05:21:54 (352 KB/s) - ‘issues’ saved [35025]
> 
>      Wireframe38.pdf already exists, exiting...
>    --
> 
>  ✓  magazine wireframe all
> 
> 9 tests, 2 failures
> ```

```

sydneyaldo, 2020-06-01:
I'll try to fix this bug, I think it's good for practice

Neethu-Mohan, 2020-06-05:
It works now.
[image]

closed 2020-06-11 by Neethu-Mohan