treehouses/cli #1407: `treehouses magazine` no Internet found error

law-lin, 2020-06-03:
None of the commands work, but I am clearly connected to WiFi.
```
root@treehouses:~/cli# treehouses magazine
Error: no Internet found
root@treehouses:~/cli# treehouses detect wifi
true
root@treehouses:~/cli# treehouses magazine
Error: no Internet found
root@treehouses:~/cli# treehouses magazine magpi
Error: no Internet found
root@treehouses:~/cli# treehouses networkmode
wifi
root@treehouses:~/cli# treehouses networkmode info
essid: NETGEAR75, ip: 192.168.1.20, has password

```

jayala-29, 2020-06-04:
not sure why this is happening, hopefully it's fixed with the updated `treehouses magazine` y'all have been improving :)

kjong, 2020-06-04:
I'm not able to reproduce this issue.

@law-lin Can you try running `treehouses internet`?

law-lin, 2020-06-04:
Closing this as it seems to work now. `treehouses internet` returned true as well at the time. I made no changes whatsoever though. Maybe my RPi couldn't communicate with https://magpi.raspberrypi.org/issues at the time?

closed 2020-06-04 by law-lin