treehouses/cli #2139: `treehouses services gitlab`?

dogi, 2021-03-25:
https://hub.docker.com/r/ulm0/gitlab

dogi, 2021-03-25:
rpi with 4gb necessary

JLKwong, 2021-04-02:
Successful Docker run command:
> `docker run -d --hostname 192.168.86.47 -p 443:443 -p 80:80 -p 2222:22 --name gitlab -v /srv/gitlab/config:/etc/gitlab -v /srv/gitlab/logs:/var/log/gitlab -v /srv/gitlab/data:/var/opt/gitlab --privileged ulm0/gitlab`

JLKwong, 2021-04-02:
First run time: 8 minutes to start up.

JLKwong, 2021-04-02:
Trying to access GitLab before it fully starts up:
[image]

JLKwong, 2021-04-02:
Successfully accessing GitLab after it finishes starting up (8 minutes):
[image]

JLKwong, 2021-04-03:
Default admin account username is `root`:
[image]

JLKwong, 2021-04-03:
By default, GitLab requires admin approval for newly registered accounts:
[image]

JLKwong, 2021-04-03:
The GitLab admin can disable the approval requirement for newly registered accounts:

[image]
[image]

rjpadilla, 2021-04-05:
> 
> 
> rpi with 4gb necessary

[image]
Wish me luck 👍 👍 👍

commit b8b6536d (2021-04-04): add services gitlab (fixes #2139)
commit d82e1c5a (2021-04-05): first testable gitlab services (fixes #2139)
commit ffeb4d35 (2021-04-05): first testable gitlab services (fixes #2139)
commit 081eb759 (2021-04-05): first testable services gitlab (fixes #2139)